SunDevilRocketry/mod #12: GPS Testing and Development

niekky, 2025-01-25:
Test GPS-Development branch with an antenna. Test is successful when there is a NMEA message with reasonable longitude, latitude, and GPS built in time data.

After the functional test, come up with a testing plan to effectively test the GPS readouts. (For instance, GPS readouts test at the desert)

niekky, 2025-02-12:
Test result (Feb 11):

GPS with patch antenna works but inconsistently, requires the patch side to point upward to collect data.
- GPS data byte was collected by gps's rx_buffer as expected.
- gps_data struct brought from rx_buffer contains very small number.
-> GPS message parser needs more testing for more accurate validation.

[image]
Figure of GPS messages collected with the patch antenna.

ETSells, 2025-02-17:
> Test result (Feb 11):
>
> GPS with patch antenna works but inconsistently, requires the patch side to point upward to collect data.
>
> * GPS data byte was collected by gps's rx_buffer as expected.
> * [...]

I'm loving the message you have highlighted. Every field I can validate looks pretty spot on. Same with all the other GGA messages. Looking forward to finding out more about our next steps tonight, this looks really promising.

closed 2025-03-04 by niekky