SunDevilRocketry/mod #21: GPS Testing

ETSells, 2025-03-04:
Issue: GPS firmware is still new and potentially unpredictable. Validate functionality via unit testing and on-the-board manual testing.

ETSells, 2025-03-04:
My specific portion: Validate GPS_parse. I'd like to specifically do this with Unity to build my knowledge of the tool as we start to use it more in other areas.

ETSells, 2025-03-04:
Bug identified in GPS_parse: when a NMEA sentence has a character field that is empty, the parser's index increments by two.

Fix: Add function in gps.c that checks if there is a character at the current index. t: parse as before. f: increment index by 1 and continue.

ETSells, 2025-03-05:
Full statement coverage achieved in GPS unit tests.

ETSells, 2025-03-06:
@journeyhancock What's your status with hardware testing? The part I was looking at is done, but if you have more you're looking at I'm willing to help out!

journeyhancock, 2025-03-06:
> [@journeyhancock](https://github.com/journeyhancock) What's your status with hardware testing? The part I was working on is done, but if you have more you're looking at I'm willing to help out!

Sorry, I actually haven't been able to do anything for this yet cause i've been working on sdec-api stuff. I don't know yet if i'll be in the lab tomorrow because of midterms but I will for sure be on Friday so i'd be more than happy to start doing my part for this. I honestly don't remember exactly what I need to work on for this so it'll take a bit, if you have any guidance for where I should start that would be greatly appreciated.

ETSells, 2025-03-06:
Okay, that all sounds good! I won't be easily reachable on Friday so I'll give you as much as I can now. @niekky if you also have anything to weigh in on, I think that'd be valuable.

The most up to date branches for everything have changed. Make sure you're on the right branches and that you've pulled up.
- Flight-Computer-Firmware: Canard-Development (GPS-Development should still work)
- mod: main
- lib: main

If Nick's sensor poll in SDEC works, that's probably the best way to test. He'd have to tell you about it since I haven't had the chance to get too far into it.

Otherwise, using GDB on app/terminal/rev2 should be fine. Unfortunately, we aren't 100% done with development, so there's a change you'll need to make for testing (I might try to see if I can get this working tomorrow, we'll see). Currently, the interrupt handler in the terminal doesn't do anything when it receives GPS data. You should add some code that takes the received buffer, and passes it into GPS_Parse( &gps_data, gps_buffer_ptr ). Hopefully that way, you'll be able to just set breakpoints and "p gps_data" wherever you want.

Also just letting you know since I didn't know lol, the orange side of the antenna is up. It's really sensitive and really bad, so you'll need to go outside and point the thing up to get any usable data.

journeyhancock, 2025-03-06:
Thanks for all the info. I'll try and get working on that after my midterms.

closed 2025-03-13 by niekky