SunDevilRocketry/mod #28: Test lora_init and lora mode set

niekky, 2025-03-26:
GOAL: 
- SPI protocols work as expected without failures
- LoRA Configuration such as Spreading Factor, Error coding, Bandwidth, LoRA mode bit are set up and written to the lora operation register.
- Lora init should initialize the module with pre defined lora_config struct

niekky, 2025-03-26:
LoRa Init is tested successfully. The driver is able to set up with parameters like spreading factors, bandwidth, and radio frequency.

niekky, 2025-03-27:
Here are test results:

Lora is initialized with this configuration
[image]

This is the sequence of commands to read registers to validate if they're written by the init function
[image]

1/ OP Mode
[image]
[image]
137 -> 1000 1001 
=> Long Range Mode: Lora Mode
=> Low Frequency Mode is On
=> Mode: Standby

2/ Bandwidth, Coding Rate, Implicit Header Mode, Spreading Factor
[image]
[image]
[image]

ModemConfig1 is 3 => 0000 0011
=> BW is 7.8 kHz
=> Coding rate is 4/5
=> Implicit Header Mode

ModemConfig2 is 96 => 0110 0000
=> Spreading Factor of 6

3/ Frequency
[image]
[image]
We need to find Frf with the desired frequency (915Mhz). There are 3 registers to write Frf: LSB, MidSB, MSB
MSB = 192
MidSB = 228
LSB = 0

=> Frf = 192>>16 + 228>>8 = 14991360 
freq = 14991360 * 32Mhz / 2^19 = 915 Mhz

ETSells, 2025-03-28:
@266-750Balloons I recommend you review these results due to subject matter expertise

closed 2026-01-15 by ETSells