SunDevilRocketry/mod #43: Improve and integrate sensor poll into canard and data-logger

niekky, 2025-04-18:
Goal: Pull out the data that we don't want to write to flash before writing to memory. This will increase flash performance and reduce code execution time.

We can pass sensor_data struct and sensor_ids params to sensor poll to return sensor data with corresponding sensor ids. This will avoid overhead and extra processing functions from sensor dump and increase code performance.

From development side, once we leverage sensor poll, sensor dump can still be kept in the code for other general purposes. Sensor poll will be used for critical applications meanwhile sensor dump will be used for debugging and new sensor data experimenting.

niekky, 2025-04-21:
Apr 20: 
Add VELO_X, VELO_Y, VELO_Z cases to sensor poll. 
Create sensor_fusion.c and .h in the sensor folder to contain all sensor calculation.
Split sensor_imu_velo into each axis velo calculation function
Split sensor_body_state into each angle unit calculation

ETSells, 2025-04-21:
Setting as target for upcoming iteration

ETSells, 2025-08-28:
Closing as won't do. A configurable implementation exists in APPA (for the flash side at least) and this is a "nice-to-have" change for the legacy applications.

closed 2025-08-28 by ETSells