SunDevilRocketry/mod #54: Convert the drivers for I2C sensors (baro and IMU) to use IT instead of blocking functions

ETSells, 2025-07-02:
Discussed with @266-750Balloons in the meeting, and I hope the description is self explanatory. If not, Nick is probably a better point of contact on this than me.

ETSells, 2025-07-02:
Don't worry about app-level design for now on this one, the focus is just the drivers. We'll create another task for handling synchronization of interrupts when this is complete.

266-750Balloons, 2025-07-30:
This is mostly where I've been doing my work, if you want to look at my progress on this:
https://github.com/SunDevilRocketry/mod/tree/ds/baro-it

ETSells, 2025-08-27:
Feature postponed to next release.

ETSells, 2025-09-17:
@266-750Balloons has made some progress here for the next assignee, if you want to coordinate.