SunDevilRocketry/mod #66: Mod: Test LoRa functionality and port to APPA

266-750Balloons, 2025-09-09:
At the end of the Spring semester, me and @niekky got basic transmission and reception working, albeit with some bugs and finickiness. Now that summer's over, I want to get the driver polished and make sure it works well with the APPA branch as a solid basis on which to start development.

Tasks:
- Test and debug the library at the workshop; I'll be leading this, but I need a partner to help with testing. This will involve using a USB software defined radio and the gdb debugger, which I can show you the ropes of.
- Update the APPA lib to support LoRa chips; this involves using STM32CubeMX software to set up pin assignments and chip clocks, and then putting the results into lib. I've done this successfully before on another branch as well, so I can give those files for references on what the correct values should be. I can teach you to do this if you want.

ETSells, 2025-09-17:
2nd assignee needed urgently

ETSells, 2025-09-17:
@266-750Balloons opened for this sprint. High priority, high difficulty.

ETSells, 2025-11-05:
The scope of this task has been broken down. See the epic for a new task devoted to work on the testbed environment.